cablelabs/lpwanserver #53: protocolHandlerNetworkFields modifications

SteveAtSentosa, 2018-07-30:
for GET .../networkProtocols 

Align prop names in `{metadata : { protocolHandlerNetworkFields }}` to align with attribute names used by HTML input elements

So, using this as reference:

[image]

### Prop renames
```
fieldName -> name         // used to name the corresponding JSON prop

fieldType -> type         // HTML input type: text, number, checkbox, email, etc

fieldDesc -> description  // Informative sentence displayed below the input field
                          // `The name of the remote IoT network` in above example

fieldHelp -> help         // I envision an info button that leads to a descriptive popup, 
                          // for example instructions RE how to register a TTN client
                          // This is a nice to have, but we can live without
```
### Prop additions

```
label                     // Label displayed above the input field
                          // `Network Name` in the above example

value                     // Initial/default value.  Can be ommited.

placeholder               // Hint text displayed in an empty input field
                          // `e.g. 'Kyrio LoRa` in the above example
```

JimCAtSentosa, 2018-07-30:
I thought we had a label value.  But yeah, "field" in everything is a little redundant again.  Regarding the help field, wondering if this might be outside the scope of the fields (that is, a level "up" in scope), rather than repeating for each field.

Also, given recent conversations, I originally proposed "displayWithQueryParameter" as a field that was a query parameter to the UI page.  Since we can't specify unique parts of the URL to the OAuth process, my thinking has changed to having some localStorage keep track of the network that we're doing OAuth on, store the data in the database when the OAuth callback happens, and then redirect back to the edit page.  So maybe a better name is just "oauthQueryParameter".

I mean, as long as we're making changes anyway...

dschrimpsher, 2018-07-30:
Works for me

closed 2018-07-31 by dschrimpsher