cablelabs/lpwanserver #167: Clicking Enable on network page causes error boxes.

dschrimpsher, 2018-10-10:


dmalas, 2018-10-31:
Two Errors (red boxes) are generated in the UI: "TypeError: Failed to fetch" and "Failed to fetch"

dmalas, 2018-10-31:
This error seems to also create a major failure condition with the server. When clicking on "Home" at the top, it generated 5 error boxes all: "TypeError: Failed to fetch"

dmalas, 2018-10-31:
This bug appears to crash the server: "docker_rest_1 exited with code 1, docker_ui_1 exited with code 1"

dschrimpsher, 2018-11-08:
So this wasn't really an enable bug.  The problem is in #199 where the redis connections which hold device activation data are incompatible between V1 and V2 and so when I update V1 after adding a device to V2 it blow everything to hell.  The problem with the enable was that the enable checkbox was causing the network to be “re-pulled” and re-pushed every time.  I fixed that part and created an issue for the redis bug.

closed 2018-11-08 by dschrimpsher